hsaturn/TinyMqtt #8: Arduino Portenta Potential?

hpssjellis, 2021-06-04:
This is really cool.

Any chance this could work on the new Arduino Portenta MBED capable board?

hpssjellis, 2021-06-09:
@hsaturn 

Looks like it is kind of working, randomly on the Portenta. The two errors coming up are:

```

Exception in thread "AWT-EventQueue-0" java.util.ConcurrentModificationException
	at java.util.LinkedList$ListItr.checkForComodification(LinkedList.java:966)
	at java.util.LinkedList$ListItr.next(LinkedList.java:888)

and

Exception in thread "Thread-258" java.util.ConcurrentModificationException
 lots more errors ...

```

Which appears to be something to do with lists of objects being changed while iterating.  Since the MQTT broker works fine on my Node-MCU-32S it must be something to do with how I am interacting with it using the Portenta.

Note: Debug ```ESP.getFreeHeap()``` kills the program but that is an easy omit.

Any suggestions for the two errors? or area in the .cpp file you think it might be happening?

hsaturn, 2021-06-09:
> This is really cool.
> 
> Any chance this could work on the new Arduino Portenta MBED capable board?
>

Hello Jeremy

Unfortunately, I'm totally unaware of the Portenta MBED board. I've left the arduino world in favour of Esp's because they are more capable and have wifi. I probably won't change again for the Portenta board ...

hsaturn, 2021-06-09:
> 
> 
> By the way, amazing code. Runs great on my NodeMCU-32S.
> 
> Questions @hsaturn
> 
>     1. What problems would you expect if I try to get it running with an Ethernet board?
>

Obviously we'll encounter some compilation problem because the code uses Wifi libraries. 
For example TinyMqtt.cpp uses for example WifiClient (that comes from EspAsyncTCP or Wifi.h)

> 
>     2. Where was your inspiration? I have not found many other small MQTT examples even in Python.
> 

Comes from nowhere, [I've just read the specifications there ](url).

>     3. If I get it working on the Portenta (with Arduinos help) would you be interested in a PR, or should I just make a separate library? My fork and re-named downloadable library `tinyMQTTPortenta` is at https://github.com/hpssjellis/TinyMqtt
> 

I'd like to say "YES feel free to send a PR" but ...

1/ I have some AUnit tests that are ESP related so : 
   1.1/ I wrote a kind of EspMock so you'll have to write a Arduino mock.
   1.2/ We should add the Arduino automatic test so it can 
   
2/ I won't be able to maintain the Arduino specialization part of the code.

3/ We may have to remove the coupling between the network layer and the mqtt layer and propose a better architecture. 
  I probably will not accept to introduce some virtuals for that (maybe templates) because I may want to keep the code fast and simple. I'm pretty sure that templates can do the job.

>     4. My Arduino core MBED issue of this is filed at
>        [arduino/ArduinoCore-mbed#251](https://github.com/arduino/ArduinoCore-mbed/issues/251)

Hummm... reading this, it seems you only had to modify the #if sequence in TinyMqtt.h
Sounds great !

Feel free to send a PR. I'll do my best to accept it.

hpssjellis, 2021-06-09:
Probably should move these to the discussion area.

```
2. Where was your inspiration? I have not found many other small MQTT examples even in Python.
Comes from nowhere, I've just read the specifications here .    https://github.com/hsaturn/TinyMqtt/issues/url

```
1. Can you repost the specifications link you used. It seems broken. 

2. Kudos to you making this library from scratch, that is way above my ability level. I just  teach average High School  students the smart things you good coders make. What you have done here is way above my ability level. Well done. 

3. I will try to get the Portenta working using tinyMQTT, probably a bad idea to accept a PR, mixing the two methods may prove very confusing. 

Lets hope the Portenta stuff doesn't break your code. Lets see how messy my code ends up. If we end up with two libraries, I can give you all the credit, I just want it working.

hsaturn, 2021-06-10:
Sorry for the link

[Here is the good one](https://docs.oasis-open.org/mqtt/mqtt/v3.1.1/mqtt-v3.1.1.html)

hsaturn, 2021-06-10:
> I will try to get the Portenta working using tinyMQTT, probably a bad idea to accept a PR, mixing the two methods may prove very confusing.

--> If this is only a "using" thing, there is no problems. As I said I'll do my best to accept a PR but obviously won't accept if this make things too confusing.

And of course feel free to fork and modify. This code is released under the GPL 3.0 license.

hpssjellis, 2021-06-12:
I am going to close this for a while until I have a better C++ compiler working. Thanks so much for spending the time to answer my questions. Most of the changes to get tinyMQTT working on the Portenta should be able to be done in the sketch. I will submit one more issue about the ```ESP.getFreeHeap()```  in the main .cpp  file

closed 2021-06-12 by hpssjellis