hsaturn/TinyMqtt #11: ESP.getFreeHeap() kill debug on non-esp cores

hpssjellis, 2021-06-12:
Please allow some way to ignore ```ESP.getFreeHeap()``` in TinyMqtt.cpp line 397 [here](https://github.com/hsaturn/TinyMqtt/blob/90dea36ab07090c7b6aa137c1a9c80e1628127ba/src/TinyMqtt.cpp#L400)    as it kills using debug on a non-esp board. 

My suggested method would be, but you probably have a better method.

was:
```

#ifdef TINY_MQTT_DEBUG
if (mesg->type() != MqttMessage::Type::PingReq && mesg->type() != MqttMessage::Type::PingResp)
{
	Serial << "---> INCOMING " << _HEX(mesg->type()) << " client(" << (dbg_ptr)client << ':' << clientId << ") mem=" << ESP.getFreeHeap() << endl;
	// mesg->hexdump("Incoming");
}
#endif

``` 

to something like:

```

#ifdef TINY_MQTT_DEBUG
if (mesg->type() != MqttMessage::Type::PingReq && mesg->type() != MqttMessage::Type::PingResp)
{
 #ifdef NOT_ESP_CORE
	Serial << "---> INCOMING " << _HEX(mesg->type()) << " client(" << (dbg_ptr)client << ':' << clientId << ") mem=" << " ESP.getFreeHeap() "<< endl;
#else
	Serial << "---> INCOMING " << _HEX(mesg->type()) << " client(" << (dbg_ptr)client << ':' << clientId << ") mem=" << ESP.getFreeHeap() << endl;
#endif

	// mesg->hexdump("Incoming");
}
#endif

```

That change will hopefully allow other boards to get this important library running.

hsaturn, 2021-07-18:
Hi Jeremy, sorry for the very late answer. I was in holidays.

I'll take care of this very soon but not today :-)

Thanks for your report.

hsaturn, 2021-08-04:
Sorry, I'm very busy these days, I'll take care of it soon

hsaturn, 2021-08-09:
Hi Jeremy

It seems that this function is used only when DEBUG mode is enabled (TINY_MQTT_DEBUG).

But as far as I know, this variable is not set:
TinyMqtt.h: // #define TINY_MQTT_DEBUG

So, either I'm wrong, or you have set this define in order to have some serial debug info ?

hpssjellis, 2021-08-09:
yes @hsaturn I was trying to get any debug information that was available. Sorry if that was confusing. 

BTW, looks like Arduino is working on being able to use tinyMQTT on the Portenta.

hsaturn, 2021-08-09:
Ok Jeremy, so I have a better understanding on how you got there. This is not a global problem.

I'll add your suggested modification in the main branch (but won't create a release for that).

Best regards

closed 2021-08-09 by hsaturn