hsaturn/TinyMqtt #13: Incorrect message length calc

Kirlik, 2021-07-29:
It looks like the length should be calculated this way:
		case Length:
			size += (in_byte  & 0x7F) *size_mult;
			size_mult *=0x80;
.............

size_mult  should initialize with "1"  in FixedHeader section

hsaturn, 2021-08-02:
Thanks for the report, I'll have a look this evening.

hsaturn, 2021-08-04:
Sorry, I'm very busy these days, I'll take care of it soon

hsaturn, 2021-08-09:
Hi Kirlik

Thanks for your report, the 3F has been fixed to 7F [here](https://github.com/hsaturn/TinyMqtt/commit/56a2be621f446cf8f0801e9a81cdc19cc42dbcc5).

Unfortunately, I cannot write a unit test for some reasons but the bug should be fixed.

Best regards

closed 2021-08-09 by hsaturn

Kirlik, 2021-08-09:
I think it's still wrong. The message length comes with big-endian. Consider using my code.

hsaturn, 2021-08-09:
Hello Jeremy

Thanks for making me look again at the specification, and you're right, then I'll fix that very soon

hsaturn, 2021-08-09:
I do not want to handle > 16k messages. I consider that this is too long for a embedded device.
I cannot see yet a use case where > 16k msg could be of any interrest for an Esp.

Feel free to comment if you think this is really to low.

Best regards

Kirlik, 2021-08-10:
But at least a couple of thousand bytes would be nice to process. For example there are packets with a length of more than 1 byte in the "tasmata" firmware for Sonoff relays  

> 
> 
>I do not want to handle > 16k messages. I consider that this is too long for a embedded device.
>I cannot see yet a use case where > 16k msg could be of any interrest for an Esp.
>Feel free to comment if you think this is really to low.
>Best regards
 
 
С уважением,
Кирилл Петюшин

hsaturn, 2021-08-10:
I do agree !

This is why I've set the limit at 4096 bytes (see [there](https://github.com/hsaturn/TinyMqtt/blob/main/src/TinyMqtt.h#L67)) in the last release.

I think that 4K allows to handle most of situations, including the 2000 you mentioned.
The hard limit is 16k, I consider that more than 16k on ESP which have 96K of memory is fair.