hsaturn/TinyMqtt #17: Broker edit JSON payload

pruwait, 2021-08-29:
My PubSubClient send to mosquitto broker topic Light/relay/json_status with payload:
{"r1":1,"r2":1,"r3":1,"r4":1,"r5":1,"r6":1,"r7":1,"r8":0,"r9":1,"r10":0,"r11":0,"r12":0,"r13":0,"r14":0,"r15":0,"r16":0,"r17":0,"r18":0,"r19":0,"r20":0,"r21":0,"r22":0,"r23":0,"r24":0,"r25":0,"r26":0,"r27":0,"r28":0,"r29":0,"r30":1,"r31":0,"r32":1,"freeheap":272172,"uptime":600}
Mosquitto and another client sees same.

If my PubSubClient send this payload to  [TiniMQTT broker](https://github.com/hsaturn/TinyMqtt/blob/main/examples/simple-broker/simple-broker.ino) and another client (MQTT Explorer, MQTT Spy) sees other:
{"r1":1,"r2":1,"r3":1,"r4":1,"r5":1,"r6":1,"r7":1,"r8":0,"r9":1,"r10":0,"r11":0,"r12":0,"r13":0,"r14":0,"r15":0,"r16":0,"r17":0,"r18":0,"r19":0,"r20":0,"r21":0,"r22":0,"r23":0,"r24":0,"r25":0,"r26":0,"r27":0,"r28":0,"r29":0,"r30":1,"r31":0,"r32":1,"freeheap":272172,"uptime":600}�

What is the ending symbol?

pruwait, 2021-08-29:
remark:
mosquitto broker send to client multi-line JSON payload, for example:
{
  "r1": 1,
  "r2": 1,
  "r3": 1,
  "r4": 1,
  "r5": 1,
  "r6": 1,
  "r7": 1,
  "r8": 0,
  "r9": 1,
  "r10": 0,
  "r11": 0,
  "r12": 0,
  "r13": 0,
  "r14": 0,
  "r15": 0,
  "r16": 0,
  "r17": 0,
  "r18": 0,
  "r19": 0,
  "r20": 0,
  "r21": 0,
  "r22": 0,
  "r23": 0,
  "r24": 0,
  "r25": 0,
  "r26": 0,
  "r27": 0,
  "r28": 0,
  "r29": 0,
  "r30": 1,
  "r31": 0,
  "r32": 1,
  "freeheap": 270620,
  "uptime": 63
}

TiniMQTT broker send one-line payload with adding simbol.

hsaturn, 2021-09-01:
Hello, thanks for the report.
Obiously, the last character is a random char added.
A reason could be that TinyMqtt pretend that the payload is one more byte than it is.

I'll check that very soon.

Thanks for the detailed report.

hsaturn, 2021-09-01:
Last bug I've fixed was about long messages length decoding...

Funny thing, I've opened an issue : "Write unit test for length message decoding" ;-)

hsaturn, 2021-09-08:
Hello

Just to tell that I keep in mind that issue. I'm very sorry for the long delay to fix this bug. I was very busy last weeks.
I think I can take care of that issue this evening

Best regards

hsaturn, 2021-09-10:
Hello

I've made severall tests. I've modified the library a little so that I can view Hexa dumps of what is sent / received.

I did not saw any problems with small payloads.
(But I've found other bugs :-))

I'm going now to make some tests with a longer payloads.

[image]

hsaturn, 2021-09-10:
And with a long payload, I can see now the spurious char

[image]

hsaturn, 2021-09-14:
Hi Pruwait,

I thinkj I have a fix for this bug. I have to do some more tests because I have rewritten many part of the code (poor code on this part...).

hsaturn, 2021-09-17:
Back at work on this problem.

hsaturn, 2021-09-17:
My fix is working in one situation (yours).
But unfortunately, I have some unit tests that are broken. So I can't release a new version yet.

hsaturn, 2021-09-17:
Hello, back again
I've released the version 0.8.0 of TinyMqtt that should fix your problem.

Feel free to reopen this issue if you still have some problem.

I wrote a unit test that is now passing successfully.
I did not test the release on MqttBox I'll do that tomorrow when the release will be available directly in the Ardiuno IDE

Best Regards.

hsaturn, 2021-09-17:
I'm reopening this topic.
The bug is still there.
The unit test is not going though any network interface. Message decoding is broken.

hsaturn, 2021-09-19:
Hello, I'm verryyyy busy working on this issue.
You may think this is a basic issue. And probably it is.
But as I really want TinyMqtt to be a quality library, I do not want to quick fix this bug.

Instead, I want to write a unit test that fails, fix it and view the unit test green.
This way, I'll be pretty sure that TinyMqtt will not suffer of regressions later.

Such a test has been written when clients do not send via wifi (but instead to another client of the same esp).
But in order to write a unit test that sends through wifi, I need to emulate part of the ESPWifi library.

This is why this bug is long to be fixed. Other bugs will be faster but I have to finish the ([EspMock](https://github.com/hsaturn/EspMock) for wifi.

Yet, I'm working on unit tests of .... esp mock, the lib that will allow better unit tests of TinyMqtt.

... Thousands of line of code for... probably two or three buggy lines :-D

Best regards

hsaturn, 2021-09-20:
Hi all, some news.

I think I've finish the tcp/ip emulation.
Basic tests with TinyMqtt are working with EspMock through virutal network.

I wrote a unit test for hudge payload (network_one_client_one_broker_hudge_publish_and_subscribe_through_network).
But unexpectedly, this test passes !!!

That can mean three things
1 - EspMock (the virtual network) has any bug
2 - The test is not written correctly
3 - I was wrong when I though my bugfix was bad on TinyMqtt
4 - All is working as expected, I mean TinyMqtt is able to communicate to TinyMqtt with hudge payloads, but when we introduce some external Mqtt messages, things are starting to going wrong.

My intuitive guess is 4).

I'll check that tomorrow. 

If you want to help me, You can clone the latest version from github (dfd5983715aaa...) and test it under your conditions.

Best regards

terrorsl, 2021-09-25:
Hi, this code is not pretty, but it works for me. Hope it helps you.  
``` c++
int sMQTTMessage::encodeLength(char* msb, int length) const
{
	int count = 0;
	do
	{
		unsigned char encoded(length & 0x7f);
		length >>= 7;
		if (length) encoded |= 0x80;
		*msb++ = encoded;
		count++;
	} while (length);
	return count;
};
sMQTTError sMQTTMessage::sendTo(sMQTTClient *client,bool needRecalc)
{
	if (buffer.size())
	{
		if (needRecalc)
		{
			//debug("sending " << buffer.size() << " bytes");
			char *bytes = (char*)buffer.data();
			char size[4];
			size[0] = bytes[1];
			if (encodeLength(size, buffer.size() - vheader/*2*/) > 1)
			{
				buffer.insert(buffer.begin() + 2, size[1]);
			}
			buffer[1] = size[0];
		}
		// hexdump("snd");
		client->write(buffer.data(), buffer.size());
	}
	else
	{
		//debug("??? Invalid send");
		return sMQTTInvalidMessage;
	}
	return sMQTTOk;
};
```

conaito, 2021-11-19:
@ terrorsl Help it to can get longer messages with your code? I try to add it into the cpp code but looks like need some more code adding (may also in header file?) Can you give me small idea to use it also? Thx for your help

terrorsl, 2021-11-19:
> @ terrorsl Help it to can get longer messages with your code? I try to add it into the cpp code but looks like need some more code adding (may also in header file?) Can you give me small idea to use it also? Thx for your help

What say compiler?

conaito, 2021-11-20:
may you can provide your complete h and ccp file? i think this is much faster :)

terrorsl, 2021-11-21:
> […](#)

look here:
https://github.com/terrorsl/sMQTTBroker 
sMQTTMessage.h and sMQTTMessage.cpp

conaito, 2021-11-21:
very cool. thx! there no problems with longer messages?

terrorsl, 2021-11-21:
> […](#)

It no problem for me, but need test.

conaito, 2021-11-21:
i will for sure. is it based on this mqtt code or complete different ?

terrorsl, 2021-11-28:
some of the code are the same, but a lot is different.

hsaturn, 2022-10-30:
Hello

The last git version (not yet the last release) should have fixed this.
Best regards.

hsaturn, 2022-10-31:
A unit test was added long ago (network_hudge_payload).
TinyMqtt accept 32767 bytes payload (I guess)
Test is on 210 bytes

closed 2022-10-31 by hsaturn