hsaturn/TinyMqtt #18: Question - How many client handle it? (ESP32-LAN)

vampywiz17, 2021-09-16:
Hello there,

What do you think, how many devices able to handle this thing a similar hw?

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

hsaturn, 2021-09-17:
Imho, this hw could handle something between 50 and 100 clients.
I've assumed that this device has 500k ram.

Also, it is planned than TinyMqtt will a able to handle a tree (mesh) of brokers / clients
(client -> broker node -> ... -> final broker)
This way, each node has to handle ~5 clients, the final broker will be able to handle hundreds of clients depending only on how many topics are used.
Even a ESP8266  could handle so many clients this way.

hsaturn, 2021-09-17:
I'm closing this issue because that is not a issue, please continue this in "Discussion". That is what it is.

Best regards

closed 2021-09-17 by hsaturn