hsaturn/TinyMqtt #24: zeroconf

conaito, 2021-11-15:
Any idea when zeroconf will be implemented? Its sounds like a great feature!
Also, are there a chance to get websockets supported?
How much clients (on broker) can currently supported/handle by a esp32 ?

valientegaston, 2022-02-14:
I wanna know too :)

hsaturn, 2022-10-30:
Hello

I'm back on TinyMqtt after some months.

First of all I've to fix some reported bugs. Next I will be able to spend time on zeroconf.

I have no plan for zeroconf yet, even if I'd like it to be implemented. But I keep this in mind.

hsaturn, 2022-11-21:
Still thinking about this. But things had to be done.
Recently wildcards has been added to the broker.

Regards

AlesSt, 2022-11-30:
As far as I ve read stuff max clients should be 16 but lwip supports max 10 meaning 9 clients + 1 socket needs to be opened for listening for new clients. So it should be max 9 clients.