hsaturn/TinyMqtt #26: Fail connect ESP8266 client

fazerlab, 2022-02-08:
Tried example simple_broker with a client esp8266 using lib pubsubclient [https://github.com/knolleary/pubsubclient](url)
and connection to broker failed. 
this the code:

``` #include <ESP8266WiFi.h>
#include <PubSubClient.h>

// Update these with values suitable for your network.

const char* ssid = "-----";
const char* password = "-------";
const char* mqtt_server = "192.168.1.116";

WiFiClient espClient;
PubSubClient client(espClient);
unsigned long lastMsg = 0;
#define MSG_BUFFER_SIZE  (50)
char msg[MSG_BUFFER_SIZE];
int value = 0;

void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void callback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  // Switch on the LED if an 1 was received as first character
  if ((char)payload[0] == '1') {
    digitalWrite(BUILTIN_LED, LOW);   // Turn the LED on (Note that LOW is the voltage level
    // but actually the LED is on; this is because
    // it is active low on the ESP-01)
  } else {
    digitalWrite(BUILTIN_LED, HIGH);  // Turn the LED off by making the voltage HIGH
  }

}

void reconnect() {
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str())) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("inTopic");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      delay(5000);
    }
  }
}

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  unsigned long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, MSG_BUFFER_SIZE, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
} 
```

Message: 
Attempting MQTT connection...failed, rc=-1 try again in 5 seconds

the simple_broker is in a ESP32. I can connect with MQTTX app https://mqttx.app/

thank you

xristiaan, 2022-02-08:
Hi,

first of all, great that someone took the time to create an MQTT broker!

I have installed it on an ESP32 chip and when I try to connect to it (pubsub on an ESP8266), it does not work I get state -1

To narrow down the problem, I have:
connected locally with MQTT explorer to the broker on my ESP32, which works
connected may ESP8266 with PubSubClient client to the HiveMQ public MQTT broker, which also works.
Set the messages sizes to 256 bytes on both the ESP8266 to match the messagesize on the ESP32

So it seems that there is a problem with the combination of an ESP32 with TinyMQTT and the ESP8266 with PubSubClient. 

Regards,
Chris Peters

xristiaan, 2022-02-08:
I recompiled the the code for the ESP32 with debug define for tinyMQTT; when I try to connect with the ESP8226, I see the following output:

xristiaan, 2022-02-08:
I also tried putting the MQTT broker on an ESP8266 instead of an ESP32, but was not able to establish a connection there either  (by the way the 16 you see is an additional Serial.println that I put in)

144 New client
---> INCOMING 10 client(1073674028:) mem=50144
Incoming size(22), state=4
Incoming | 10 14 00 04 4D 51 54 54  | .... - MQTT | 
Incoming | 04 02 00 0F 00 08 58 52  | .... - ..XR | 
Incoming | 49 53 4D 51 54 54        | ISMQ - TT   | 
16
Connected client:XRISMQTT, keep alive=15.
772 sending 5 bytes
Free memory 49440  
176 Client XRISMQTT  Disconnected, parent=1073669844
60 close XRISMQTT
130 Remove 1
132 Client removed 0

fazerlab, 2022-02-08:
I tried ESP32 and ESP8266 and confirm that can´t conect

valientegaston, 2022-02-14:
Hi! Same problem here.  Broker on ESP8266, Client with pubsubcliente lib on ESP12E… 

 *************** Error msg 0x10-------ERROR ------ size(47), state=4
00:11:41.125 -> -------ERROR ------ | 10 2D 00 04 4D 51 54 54  | .-.. - MQTT | 
00:11:41.125 -> -------ERROR ------ | 04 C2 00 0F 00 0D 65 73  | .⸮.. - ..es | 
00:11:41.125 -> -------ERROR ------ | 70 33 32 2D 63 6C 69 65  | p32- - clie | 
00:11:41.163 -> -------ERROR ------ | 6E 74 2D 00 0A 6D 79 63  | nt-. - .myc | 
00:11:41.163 -> -------ERROR ------ | 72 6F 63 6C 69 6D 65 00  | rocl - ime. | 
00:11:41.163 -> -------ERROR ------ | 06 70 75 62 6C 69 63     | .pub - lic  | 
00:11:41.163 -> +-- 'esp32-client-'  ON , alive=396773/391804, ka=15 connected

LemarinelNet, 2022-03-30:
To add on this, I also faced this issue.
What I've found :  the PubSubClient awaits a repsonse length of 4 bytes on connection successfull, and the TinyMQTT sends 5 bytes. I deleted the "msg.add(0)" after "MqttMessage msg(MqttMessage::Type::ConnAck);" and then the pubsub client works OK.
I suppose this is a small implementation error, since connecting with PubSubClient to a mosquitto server works fine ? What about a fix ?

Hudoba, 2022-04-30:
I have a similar problem. I running "simple-broker"
*************** Error msg 0x10-------ERROR ------ size(124), state=4
-------ERROR ------ | 10 7A 00 04 4D 51 54 54  | .z.. - MQTT | 
-------ERROR ------ | 04 E4 00 3C 00 0D 6F 70  | .⸮.< - ..op | 
-------ERROR ------ | 65 6E 65 76 73 65 2D 65  | enev - se-e | 
-------ERROR ------ | 30 62 34 00 16 6F 70 65  | 0b4. - .ope | 
-------ERROR ------ | 6E 65 76 73 65 2F 61 6E  | nevs - e/an | 
-------ERROR ------ | 6E 6F 75 6E 63 65 2F 65  | noun - ce/e | 
-------ERROR ------ | 30 62 34 00 43 7B 22 73  | 0b4. - C{"s | 
-------ERROR ------ | 74 61 74 65 22 3A 22 64  | tate - ":"d | 
-------ERROR ------ | 69 73 63 6F 6E 6E 65 63  | isco - nnec | 
-------ERROR ------ | 74 65 64 22 2C 22 69 64  | ted" - ,"id | 
-------ERROR ------ | 22 3A 22 33 30 63 36 66  | ":"3 - 0c6f | 
-------ERROR ------ | 37 32 39 65 30 62 34 22  | 729e - 0b4" | 
-------ERROR ------ | 2C 22 6E 61 6D 65 22 3A  | ,"na - me": | 
-------ERROR ------ | 22 6F 70 65 6E 65 76 73  | "ope - nevs | 
-------ERROR ------ | 65 2D 65 30 62 34 22 7D  | e-e0 - b4"} | 
-------ERROR ------ | 00 00 00 00              | .... -      | 
+-- 'openevse-e0b4'  ON , alive=32159/27249, ka=60 connected

epiller, 2022-06-10:
> To add on this, I also faced this issue. What I've found : the PubSubClient awaits a repsonse length of 4 bytes on connection successfull, and the TinyMQTT sends 5 bytes. I deleted the "msg.add(0)" [...]

This worked! Thanks @LemarinelNet

ambanmba, 2022-08-03:
Not sure if it's related, but I cannot connect with a Tasmota ESP32 to an ESP8266 running TinyMqtt as a broker. I have no problem connecting with a different client though.

*************** Error msg 0x10-------ERROR ------ size(70), state=4
-------ERROR ------ | 10 44 00 04 4D 51 54 54  | .D.. - MQTT | 
-------ERROR ------ | 04 AE 00 1E 00 0B 44 56  | .⸮.. - ..DV | 
-------ERROR ------ | 45 53 5F 33 36 42 30 34  | ES_3 - 6B04 | 
-------ERROR ------ | 30 00 17 74 65 6C 65 2F  | 0..t - ele/ | 
-------ERROR ------ | 74 61 73 6D 6F 74 61 5F  | tasm - ota_ | 
-------ERROR ------ | 33 36 42 30 34 30 2F 4C  | 36B0 - 40/L | 
-------ERROR ------ | 57 54 00 07 4F 66 66 6C  | WT.. - Offl | 
-------ERROR ------ | 69 6E 65 00 09 44 56 45  | ine. - .DVE | 
-------ERROR ------ | 53 5F 55 53 45 52        | S_US - ER   | 
+-- 'DVES_36B040'  ON , alive=370490/365536, ka=30 connected

hsaturn, 2022-10-30:
Hello to all of you

Thanks for reporting.
The bug should be fixed by the latest version (the git version). The reason is that the ConnAck packet is bad (even if well formed, the length encoding should allow what my broker sends, but this is not the case).

This version is still under dev (some more fixes).
If someone can test this git version and keep me informed.
I've added a unit test to check that ConnAck packet is 4 bytes and well formed.

Regards

hsaturn, 2022-10-31:
I've released the 0.9.3 version, which should already have fixed this issue.

closed 2022-10-31 by hsaturn