hsaturn/TinyMqtt #35: Broker Never Connects

toneware, 2022-07-30:
I used the sample broker ino file, but the broker never connects to the router. I tried several ESP8266 devices (all able to connect as mosquitto clients, FWIW), and I tried two different networks/routers. All to no avail. I'm not sure what the issue is.

hsaturn, 2022-10-30:
Hello the git version of TinyMqtt should have fixed this.
There was a problem in ConnAck message.

hsaturn, 2022-10-30:
Hello, I guess the problem is now fixed (dowload the latest release [there](https://github.com/hsaturn/TinyMqtt/archive/refs/tags/0.9.3.zip).
Note that a problem with installation from IDE may still exist so you have to install manually the library 0.9.3.
Let me know if that fixes your problem

hsaturn, 2022-11-21:
Now answer and no other complain

closed 2022-11-21 by hsaturn