hsaturn/TinyMqtt #41: Invalid CONNACK packet

cepr, 2022-10-19:
The CONNACK packet generated by TinyMqtt doesn't look right.

TinyMqtt generated this: 20 82 00 00 00
I was expecting: 20 02 xx xx (Assuming MQTT 3.1.1)

This bug prevents Adafruit_MQTT_Library from connecting to this broker. I'm sure other MQTT clients will have the same issue.

xenoeng, 2022-10-19:
This might explain why I couldn't connect from either MQTT Explorer on the PC or from MQTT Analyser on my phone. I could connect to the AP fine but had no luck with any client. Thought it was something I was missing.

cepr, 2022-10-19:
I fixed this issue by replacing:
```
MqttMessage msg(MqttMessage::Type::ConnAck);
msg.add(0);	// Session present (not implemented)
msg.add(0); // Connection accepted
msg.sendTo(this);
```
to:
```
write("\x20\x02\x00\x00", 4);
```

Unfortunately, I am getting more errors now.
I switched to [sMQTTBroker](https://github.com/terrorsl/sMQTTBroker) which works better for me.

hsaturn, 2022-10-29:
Hello cepr

For some optimisation reasons, the size with TinyMQTT is always on 2 bytes.
This is the reason for which you have 0x82 00 and not 0x02, this should not be a bug but ....

And.... for an obscure reason, the ConnAck packet does not respect the variable packet length encoding (see 2.2.3 Remaining Length).

As you've pointed this bug, I have to rewrite the packet length encoding...

Thanks for your report.

hsaturn, 2022-10-30:
Back...

I wrote a data observer in ESP8266 Wifi mock in order to add a unit test before fixing that bug.

hsaturn, 2022-10-30:
Ouch.... using the new feature of ESP8266 Mock, I can see this :

```
       |   data sent  15 : Connect     11 81 00 00 01 41 51 51 51 01 00 00 01 00 00 
       |   data sent   5 : ConnAck     21 81 00 00 00 
       |   data sent  11 : Subscribe   81 81 00 00 00 00 01 61 21 61 00 
       |   data sent   6 : SubAck      91 81 00 00 00 00 
       |   data sent  11 : Subscribe   81 81 00 00 00 00 01 61 21 61 00 
       |   data sent   3 : Disconnect  E1 81 00 
```

1 - I don't see 0x82 but 0x81
2 - I can see this everywhere !!!

Strange, nothing should work at all...
Thanks to this unit test, the fix will long last :-)

hsaturn, 2022-10-30:
Hello cepr

I've fixed the length encoding for (all) messages.
It it possible for you to check the latest git version (not released yet) so  can release the version ?

```
       |   data sent  14 : Connect     10 0C 00 04 4D 51 54 54 04 00 00 0A 00 00                   [....MQTT......]
       |   data sent   4 : ConnAck     20 02 00 00                                                 [ ...]
       |   data sent  10 : Subscribe   82 08 00 00 00 03 61 2F 62 00                               [�.....a/b.]
       |   data sent   5 : SubAck      90 03 00 00 00                                              [�....]
       |   data sent  10 : Subscribe   82 08 00 00 00 03 61 2F 62 00                               [�.....a/b.]
       |   data sent   2 : Disconnect  E0 00                                                       [�.]
```

Thanks again for reporting.

cepr, 2022-10-30:
Thanks hsaturn, I'll try now!

hsaturn, 2022-10-30:
> Thanks hsaturn, I'll try now!

Thanks to you. Keep in mind that this is the git version yet, the lib has not been released with the fix yet.

hsaturn, 2022-10-30:
Thanks a lot Cedric !!!

I will add the 3 lines you mention. Even if I do not know who else than you will use them, and even if I do not understand what they mean.

I have released the 0.9.0 version with this bugfix. But it seems now that my lib cannot be installed easilly (dependency with AsyncTCP...)

Also I have to fix a bug on incomplete decoding of topics ([Issue #30](https://github.com/hsaturn/TinyMqtt/issues/30)).

The 1.0.0 is not far, thanks to you

closed 2022-10-30 by hsaturn