hsaturn/TinyMqtt #52: Client life, local to broker, too short?

Bazmundi, 2022-12-03:
On my AP host running a tinyMQTT broker, I have the code:

```
MqttBroker broker(1883);
MqttClient peddle_client(&broker);
MqttClient pump_client(&broker);
```
The `pump_client` is acting as a proxy for the remote pump.

There is a connection topic that the pump publishes on to tell the pedal it's up.

The proxy pump client then does a few things, including resetting a count so that the debugging on both keeps peddle presses since pump reset in sync.

However, the pump client on the pedal runs out of life early and so never catches any subsequent resets of the actual pump.  I get the inaugural connection topic and any re-connections, but only up to the life expiry of the pump proxy running on the host.

I have set the default keep_life `void connect(std::string broker, uint16_t port, uint16_t keep_alive = 65535 );` to max out the uint16_t to give me 18 plus hours of connection.  This is fine for instruments I am running this on in my lab.  I won't have the instruments up for that long (a couple to 6 hours probably).

This works fine for remote clients, but does not work for local clients.  A sneaky attempt to hard-code `keep_alive = 65535` in `void connect(MqttBroker* parent);` did not work, not sure then how to max out life of local clients.

I am currently guessing the local clients are only alive for 30 seconds after creation?  Or 30 seconds topic processing?

hsaturn, 2022-12-03:
Keep_alive fixed, with a unit test added. Thanks to you.
Maybe that'll fix your problem...

Arduino IDE 2 is a pain in the A...
I've lost several changes because of the file management of this crappy IDE.
I've fixed the reboot bug !
Francois Biot

hsaturn, 2022-12-04:
Hmmpf, it seems that I made a confusion with your other issue.
*( I'm working on a esp visual debugger, because eclipse / vscode / espressif are not working for me !!! Tired of being unable to have a minimal debugger. )*

Local clients should have infinite life, keep_alive is not used for local clients. Or this is a bug. I'm gonna check this.
Thanks for the report.

alranel, 2022-12-05:
I confirm I'm seeing the same issue: local client silently stops receiving after less than a minute. I'm basically using the "client with WiFi" example and sending messages from another host using third-party clients.
(By the way, I second the request for a "buy me a beer" button :))

hsaturn, 2022-12-06:
Thanks for your feedback. I'm taking care of this.

Francois Biot

alranel, 2022-12-27:
Any progress on this? If none, I can try to help.

hsaturn, 2022-12-28:
Sorry, the end of the year seems to be unproductive... No progress on that. I'll be back soon

hsaturn, 2022-12-28:
Hello, good news, I this I have reproduced the problem.
I'm going to write a unit test in order to fix this.

Best regards

hsaturn, 2022-12-29:
Hello, back on this.

I've written some unit tests for this. These tests rely on a modification of EpoxyDuino project, (for time simulation). So I have to wait before pushing the change in main.

But in the meantime , maybe you could try the 'alive' branch of TinyMqtt and tell me if things get better ?

Here are the changes that I've made https://github.com/hsaturn/TinyMqtt/pull/55/files

alranel, 2022-12-30:
Thank you for that! I will test it by having a broker running for some days. Will keep you posted.

alranel, 2023-01-01:
So, I had a chance to test the 'alive' branch. In order to compile it with the Arduino IDE I had to inject this workaround in TinyMqtt.h because unfortunately `std::make_unique()` is a C++14 feature but ESP32 core is limited to C++11 at the moment:

```c++
#if __cplusplus == 201103L
namespace std {
  template<typename T, typename... Args>
  std::unique_ptr<T> make_unique(Args&&... args)
  {
      return std::unique_ptr<T>(new T(std::forward<Args>(args)...));
  }
}
#endif
```

Now, while testing this I'm observing this behavior:
* The callback set with `setCallback()` is never called on incoming messages (I tested subscribing to `#` but I'm getting nothing)
* I'm still able to publish messages successfully from the local client
* Remote clients get all messages correctly (tried with `mosquitto_sub`)

I don't know yet whether the keepalive issue is solved or not.

Any clue on what could be going on?

hsaturn, 2023-01-01:
Hello

What s shame. I've used Arduino IDE 2 and compilation is fine.
But with Arduino 1.8.18 event #include <memory> fails !!!

So, just with that, you helped a lot. I'm going to deal with this problem first.

Thanks a lot

hsaturn, 2023-01-01:
std::make_unique has been removed.

hsaturn, 2023-01-01:
2 - The problem with subscribing to #

The example advanced/mqtt-clas-binder wotks well (not #)
-> And you're right, not callback for # (my unit test are wrong ?)

hsaturn, 2023-01-02:
Many problems with unique_ptr.

So I've splitted clients in two parts : remote clients and local clients.
I'll be back on unique_ptr for clients, but for yet, I manually delete the clients in the destructor of MqttBroker.. Too bad.

For the # problem, not found yet but
1 - I've added a local test with subscribe to '#' which works well
2 - The advanced/mqtt-class-binder does not work.

So it seems to be a problem in class binder.

All the changes I've made are in the 'alive2' branch (not alive but alive2)

to be continued...

alranel, 2023-01-02:
Thank you for the quick feedback! It seems to be working well now.
I'm now having it run for a couple days, will let you know if the keepalive issue is solved as well. :)

alranel, 2023-01-02:
Hm, I have run the 'alive2' branch for 5 hours. At the moment it looks like the keepalive issue is still present. Messages are still routed between external clients (I see them via mosquitto_sub), but apparently the local client isn't getting messages nor is able to publish. It was working for a while when started. Any clue on how could I help you troubleshoot?

hsaturn, 2023-01-02:
Thanks a lot. You are really helping a looot.

I think there is a memory problem somewhere. When I wanted to use unique_ptr, I had some double free problems (!!!). This is very hard to see on the device, so yet, I'm trying to fix this with https://github.com/hsaturn/EpoxyDuino and https://github.com/hsaturn/EspMock with a decent debugger.
I'm gonna try to sanitize this.

I already can see that there is a problem upon client disconnection (the Broker still loops on it, it is never destroyed). So if many clients connect and disconnect, a leak occurs. Maybe during the 5 hours, external clients has disconnected some time to time.

Thanks a lot. I'll be back with a git pull soon.

hsaturn, 2023-01-03:
bug: local clients are disconnecting after 30sec when they are publishing ...

hsaturn, 2023-01-03:
bug fixed : disconnection after publish + unit test added with this commit https://github.com/hsaturn/TinyMqtt/commit/670d67c02442ae778ddd01b613a467578ddff9e4
Available on branch [alive2](https://github.com/hsaturn/TinyMqtt/tree/alive2)
... next !

hsaturn, 2023-01-03:
I've release 0.9.12 with this bugfix as all unit tests were successfull.
But there is still a big problem of memory leak with an external client that connects and disconnect a lot.
The client is not freed. But When I free it, the program crashes.
I think I'm gonna stop to try to debug this with the real device, but instead I'll try tomorrow to write a unit test for this. Debugging on an ESP is not comfortable enough.

Thanks again

alranel, 2023-01-03:
Do you recommend I test 'main' or 'alive2' now?

hsaturn, 2023-01-03:
alive2 is far from main. This branch is meant to be merged on master as soon as possible.

So if you want to test, checkout alive2.

But I have to fix the latest (and strange) bug before. Testing is right won't probably give any interresting things.
I'm still fixing things shown by your last (and great) test.

Thanks again

alranel, 2023-01-20:
Hi there! I had a chance to test again the current alive2 branch and I confirm I'm still seeing the behavior described in [my last comment](https://github.com/hsaturn/TinyMqtt/issues/52#issuecomment-1369036800), so apparently the local client disconnects after a while. How can I help you? I'd really like to make this work :)

hsaturn, 2023-01-23:
Hi, sorry, I'm working a lot on libraries that makes unit tests a lot more powerfull (and that'll be used for TinyMqtt at first in fact). This job is still in progress, but almost finished.
I want to fix this too, but things are a lot more complex that I thought, and I've found important bugs while attempting to fix the timeout thing. Those bugs needs a lot more powerfull unit test framework, so I'm working on this.
Once done, I could fix the bug, and the unit test framework will garantee that the bug is fixed forever.

hsaturn, 2023-02-02:
Hi

I've finished a hudge work: https://github.com/hsaturn/EpoxyDuino/actions/runs/4071526753/jobs/7013406246

EpoxyDuino now allows to write some unit tests with simulated time.
I this feature is not too buggy (because very new), I'll be able to write a unit test that demonstrate this issue and fix it forever.

hsaturn, 2023-02-23:
Too many noise on this issue. I'm reopening a new one https://github.com/hsaturn/TinyMqtt/issues/62.
I'm not sure that timeout problems are still exist.

alranel gave me valuable information, so I'm starting from his setup.

closed 2023-02-23 by hsaturn