hsaturn/TinyMqtt #54: Bind a class function to setCallback function

letmp, 2022-12-22:
I would like to bind the setCallback function to a class function. This seems to be impossible at the moment since MqttClient:CallBack is inaccessible. Or do I miss something and there are other ways to call a class function from the callback function?

hsaturn, 2022-12-23:
Hello

You can use lambda for this:

client->setCallback([&instance](const MqttClient* srce, const Topic& topic, const char* payload, size_t len)->void
{
   instance.onPublish(source, topic, payload, len);
});

letmp, 2022-12-23:
Thanks for your quick answer!
I already tried that, but I always get this error:

`no suitable conversion function from "lambda [](const MqttClient *source, const Topic &topic, const char *payload, size_t len)->void" to "MqttClient::CallBack" exists`

Any ideas what could cause that?
Thanks for your effort!

hsaturn, 2022-12-24:
Strange, I tried this solution before sending it to you... I remember that I had tried it with tinytest.
I'm going to write this again and send this to you.
Are you sure that you have the latest release ?

hsaturn, 2022-12-24:
What I've done:

1 - open TinyMqtt examples / tinymqtt-test
2 - at line 682 (just after client->setCallback(onPublish);
``` c++
          client->setCallback([&x](const MqttClient *srce, const Topic &topic, const char* p, size_t len)->void
          {
            x.onPublish(srce, topic, p, len);
          });
```
3 - At the begining of the file
``` c++
class X
{
  public:
    void onPublish(const MqttClient* srce, const Topic &topic, const char* p, size_t len) {}
};

X x;
```
4 - Compile fine !

Not tested, but pretty sure this'll work.
Version of IDE : 2
Version of TinyMqtt : 0.9.8

hsaturn, 2022-12-24:
Sorry, 0.9.8 was only existing on my PC, and will never exist for some reasons.
I've just released the 0.9.9.

Anyway the lambda should work with any release.
Best regards

hsaturn, 2022-12-26:
Hello Can I close this issue or is it still alive ?

letmp, 2022-12-26:
I will test it tomorrow. Thanks

alranel, 2022-12-27:
I had this problem as well. If I recall correctly, you can use a lambda here only if it does not capture. In fact, according to the C++ standard, only non-capturing lambdas can be converted to function pointers.
Maybe turning `MqttClient::CallBack` to `std::function` could be the best solution, although it would introduce some overhead.
In my case, I created my clients as global variables and created static callbacks.

hsaturn, 2022-12-28:
Hello Alranel and Letmp

@Alranel, you're right. Memory on embedded devices is the main issue we are facing. This is why I've designed TinyMqtt with a function callback instead of any other design (inheritance, std::function ...).
This is also the reason why both MqttClient and MqttBroker classes do not make use of virtual.

@letmp, As I said in my exemple with tinytest, the X instance is at the begining of the file. So it has to be global !
You cannot use a local instance of any class and put a callback on it for two reasons:

- TinyMqtt class cannot store a pointer (the local instance) in any way, even with bind because in this case, bind would need some extra bytes that cannot be stored. My example works because x is global.
- Having a callback on scoped instances is dangerous: What if the scope is destroyed if a Mqtt message arrives.

The destructor of MqttClient take care about point 2.

hsaturn, 2022-12-28:
@letmp -> (please read the previous answer)

- The callback function must be callable in all situation, or this would lead to crashes.
  So either the function has to be static of a class, or the instance has to be global as said previously.
- I will not add extra bytes to MqttClient for solving this issue.

In few words, I won't force other users of TinyMqtt to support the overload that would be needed to achieve binding to a class.

But there is a solution to move the overhead from TinyMqtt to your side:
-> The idea is that we already have a pointer on the client.

So it is possible to use this pointer to route the message to another instance.

Here is the solution for your problem
https://pastebin.com/SwaTGUyR

This solution is not fully satisfying because it can crash if the MqttClient instance is destroyed when MqttReceiver::onPublish makes use of it. Also only the last registered MqttReceiver on a given MqttClient will receive the message.

Maybe I'll add the binding class to the library (which requires to write unit tests and an example for this).

hsaturn, 2022-12-28:
Hello again

I've released the version 0.9.10 that should take care of this problem.
See the example 'advanced'.

closed 2022-12-28 by hsaturn

letmp, 2022-12-30:
Thank you very much for your effort!