hsaturn/TinyMqtt #57: AsyncTCP not working yet?

AlesSt, 2023-01-19:
Hi,

thank you very much for the library (as i believe yours is the only one that out of the box my windows clients managed to connect to and subscribe to topics as they should :) )

So back to topic :) is AsyncTCP not implemented yet? Because my clients cannot connect to ESP32 broker when adding:
`#define TINY_MQTT_ASYNC`
before including your lib

Also I have a question regarding ethernet implementation.
I'd like to use ESP32 with W5500 ethernet module with stock Ethernet lib v 2.0.1 ( Ethernet=https://github.com/arduino-libraries/Ethernet @ 2.0.1 => am using PIO on VScode) and some MQTT broker lib. And by checking your lib i guess you make only 2 calls to WiFiServer/WiFiClient (stock WiFi lib) so I guess it wouldnt be too much of a hassle to port it on wired connection.

AlesSt, 2023-01-19:
Actually I made 4 changes to your lib and stock Ethernet lib with my W5500 module is working :)

Changes I made ( if you want I can do a PR make the changes => kindda same as you with Async and I can also add an example => just let me know):
- in file TinyMqtt.h  row: 18 from: #include <WiFi.h> => #include <Ethernet.h>
- in file TinyMqtt.h  row: 60 from: using TcpClient = WiFiClient; => using TcpClient = EthernetClient;
- in file TinyMqtt.h  row: 61 from: using TcpClient = WiFiServer; => using TcpClient = EthernetServer;

- in file TinyMqtt.cpp  row: 46 from: tcp_client = new WiFiClient(*new_client); => tcp_client = new EthernetClient(*new_client);
- in file TinyMqtt.cpp  row: 173 from: WiFiClient client = server->available(); => EthernetClient client = server->available();

ALSO ( !!! VERY IMPORTANT !!! ) if you get Server.h errors while building/uploading sketch to ESP board you need to change one row in stock espressif Server.h file (the file is different than the stok arduino file):
in file Server.h row: 28 from: virtual void begin(uint16_t port=0) =0; => void begin() {};
BUT BE VERY careful with this change as IT might break the WebServer from WiFi.h (to use that you need to revert the Server.h change)
Server.h can be found on path(windows/PIO): C:\Users\xxxx\.platformio\packages\framework-arduinoespressif32\cores\esp32\  => xxxx is your username in windows (if paths in platformio hasnt been changed )

Actually I will do the PR and you can merge or reimplement in your own way.
And I need to compliment your coding style (clean and efficient)! 👍

hsaturn, 2023-01-23:
Hello,

I'm happy you've found alone the way to use the using declarations in order to switch from Wifi to Ethernet.
This is exactly why there are here. You were lucky that the Ethernet API behaves the same as the wifi one, even if these libs are made to be compatible. I'm waiting for your PR, but please take the latest dev git version, (not a release), as the code evolves quickly.

AsyncTCP should work, but I did not used it since a long time, just because I'm spending my time on bug fixes.
Async worked at the very begining. I'm pretty sure that it could work with few fixes.

The dev has stopped since 4 weeks because I'm spending my time on EspMock / EpoxyDuino (my fork) in order to create some very powerfull unit tests on github (or PC).

This issue will probably not be fixed until I finish this important work. And also fix some timeout issues (if they are still there).
I've encountered also a serious bug in some situations. And I want to get rid of std::string at max (they are taking 24bytes intead of 12 for String, but are more powerfull, because the MqttClient class seems to be too hudge (~100 bytes, that's too much for a Tiny library).

AlesSt, 2023-01-31:
I switched to PlatformIO and still have problem sometimes with internals of the IDE :) if you #define xxx it is not picked up by libs .cpp files :)    You always have to add:  build_flags = -D TINY_MQTT_DEBUG=5 ( but here you accept only 0, 1 and 2 right???) 

Well since I renamed this issue => Async I will try after I ll port lib to work with stock Ethernet lib.

Will have to check what are those(EspMock and EpoxaDuino) libs about :)

Size => I am not really too concerned about it since I guess I ll be running broker on an empty ( no other "service" will be running along broker ) ESP32 

Timings/timeouts => with changes mentioned above my W5500 module is working but clients are being disconnected by the broker on about 30 to 60 seconds. I think I noticed somwhere you had a 30s client disconnects also but you managed to solve it (at least no disconnects when WiFi lib used) -------------------> this is the biggest issue I have right now, so if you could send me to the right direction I d be VERY grateful :)

hsaturn, 2023-02-02:
Hello there is already an issue about platform. 
The version there is pretty old.

I have finished the budget work that d allow me to write an unit test for the time out issue. I want that kind of bug to be fixed once for all and never appear later in this is what are unit tests for. And this is what EspMock AUnit and EpoxyDuino forks have evolved a lot since last week's.

Thanks for the report. I lol come back right when current issue are fixed.

Best regards

hsaturn, 2023-02-23:
Hello

Late answer because I had many issue to fix (and some other open source project to maintain...)

The #define TINY_XXX is a bad idea. This does not work well.
-DTINY_MQTT_DEBUG for example is a compiler option and in fact, cannot be included in a source file.

I have to find a better way to do.

AlesSt, 2023-03-01:
Well actually I stopped porting to W5500 as I tried the LAN8720 and works with your unchanged lib :)
Lan8720 uses WiFi lib under the hood so no compatibility issues :)
With that compatibility I mainly solved my case. 
Also I saw u did some bug cleaning ... will try ur new version this week :)
as already said .... thanks for the effort and GREAT job :)

closed 2023-03-01 by AlesSt

hsaturn, 2023-03-02:
Great news !

There is a little problem, I intend to move to Async Wifi by default.
Could you try to add #define TINY_MQTT_ASYNC at the begining of TinyMqtt.h in order to see if the lib will still compile ?

Best regards