hsaturn/TinyMqtt #58: Platformio update

Marf97, 2023-01-31:
Hi,

I'm using platformio for my esp32/8266 projects and I would like to use your library for a school project :) But I would like to make use of wildcards and the version of your library on platformio is 0.8.0 (2 years old). Could you update it please? Thank you :)

Martin

Marf97, 2023-01-31:
Temporary resolved the issue by using `lib_deps = https://github.com/hsaturn/TinyMqtt.git` but it's a pitty that your library is not updated there too :/

Marf97, 2023-02-02:
What Time Out issue is happening right now?

hsaturn, 2023-02-03:
Not sure but it seems that when in local mode, local clients are disconnected after the timeout.
I'm on it this week end

Marf97, 2023-02-03:
Yes, I've been experiencing some type of timeout. I have 3 devices right now, one device creating the WiFi network, being broker and having client for listening to all messages and other two devices are just clients in the network sending their data. I haven't measured it but after some period of time, those 2 devices say they are not connected to broker. The first device managing broker, wifi network and client are working for a few days now

hsaturn, 2023-02-06:
Yes I *have* to fix that. But with a unit test so this fix will last forever.

Marf97, 2023-02-16:
@hsaturn Hello, is there any update on the issue? I could try to help out and take a look.

hsaturn, 2023-02-23:
Hello, sorry for late answer, I worked on memory leak issue.
About the timeout issue, I made some unsuccessful tests. I'm working now on a auto-reconnect feature which may leads me to this issue. Best regards

hsaturn, 2023-02-23:
Ouch !!! 0.8.0, same as yours.

I'll try to see why platform io does not update the version this evening (have to go to work now :-))

Marf97, 2023-02-23:
I guess since you opened new issue about timeout/client life, I will transfer to that one if I have any finding. I'm also working so in spare time I could take a look at it. Platformio update could be somehow automated but no experience with this yet. I think this command should be related to the version update:
https://docs.platformio.org/en/stable/core/userguide/pkg/cmd_update.html#cmd-pkg-update

hsaturn, 2023-02-23:
The funny thing is that the Changelog of platformio show the 0.9..9 release ...

hsaturn, 2023-02-23:
I've re-added a json file as it seems to be important for platformio. We have to wait for platformio to parse this file now.
The 0.8.0 release had such a file and I decided to remove it. But it seem that was a bad idea.
We will see tomorrow I hope !

hsaturn, 2023-02-23:
Problem solved

closed 2023-02-23 by hsaturn