hsaturn/TinyMqtt #61: Request for example - receive (get payload) subscribed messages

spicer1, 2023-02-20:
Dear all,
in the examples, after subscribing to a topic, I didn't find how to receive messages and "use" (get) the payload of subscribed messages.
could you add an example of receiving broadcast/posted messages from another device.

Using mosquitto for debugging I can transmit, but when I transmit messages from mosquitto with ESP8266/TinyMQTT I cannot receive, i.e. print the message.

This is the code I used... and I think I either omitted something relevant or (probably) didn't understand how the code needed to "receive" works.

I realize that the PULL request section is not the right place to ask this question... consider it as a request to update the library user guide :-)

Thanks for your kind reply.

/********************************************/
#include <TinyMqtt.h>

std::string topic_power="energy/power";
std::string topic_timing="energy/timing";

charbuffer[200];

MqttBroker broker(1883);
MqttClient mqtt_a(&broker);

void onPublishA(const MqttClient* /* source */, const Topic& topic, const char* payload, size_t /* length */)
{
   Serial << topic.c_str() << ", " << payload << endl;
}

  void setup()
{
   broker.begin();
   mqtt_a.setCallback(onPublishA);
   mqtt_a.subscribe(topic_timing);
    .....
    .....
}

void loop()
{
    .....
    .....
   broker.loop(); // Add loop for every broker and client
   mqtt_a.loop();

   static const int intervalA = 5000; // publishes every 5s (please avoid usage of delay())
   static uint32_t timerA = millis() + intervalA;

   if (millis() > timerA)
   {
      Serial.printf("mqtt_a connected:%d\n",mqtt_a.connected());
      Serial.printf("Broker connected:%d\n",broker.connected());
      timerA += intervalA;
   }
    .....
    .....
}

hsaturn, 2023-02-21:
I guess you are trying to receive your own message.

mqtt_a will not see its own message.
If you instanciate mqtt_b and subscribe then you'l receive the msg.

But i'm in hurry so maybe I have not understood (have to go to work) and have just seen this issue. Will answer / reread once back.

hsaturn, 2023-02-22:
Hello again
You are in this sketch:

- Create a local broker that is able to listen for clients and forward mqtt messages to any subscribers
- Create a local client that want to listen to "energy/timing"

```You've found a bug because the client is not connected anywhere, thanks for this !!!```

- The sketch does not open Serial (Serial.begin(speed)  (or maybe you hide it ?)
- The sketch does not open a connection to your wifi network (or did you hide it ?)
- ```The client does not connect to any broker (client.connect("somewhere")```

The broker seems useless, because you already have an external broker, (or maybe you wanted to link the broker with mosquito).

I have to fix the connected bug of client, so I leave this issue opened.

Regards

hsaturn, 2023-02-23:
Hello the example you are requesting seems to already exist: [simple-client](https://github.com/hsaturn/TinyMqtt/tree/main/examples/simple-client)

hsaturn, 2023-02-23:
(Closing this issue because no answer from spicer1 and I think I have fully anwsered.)

closed 2023-02-23 by hsaturn