hsaturn/TinyMqtt #64: Platformio fixes

hsaturn, 2023-02-23:
Check that examples under platformio compile and run.

The Arduino IDE compile options allows things that are not allowed under platformio. I have to refactor a little in order to make the compilation successful

tv4you2016, 2023-02-24:
[image]

tv4you2016, 2023-02-24:
Fix: 

```
TinyString& TinyString::operator=(TinyString && other)
{
  if (this == &other) return *this;
  clear();

  //str = std::exchange(other.str, str);
  //size_ = std::exchange(other.size_, size_);
  //free_ = std::exchange(other.free_, free_);
  using std::swap;
  swap(other.str, str);
  swap(other.size_, size_);
  swap(other.free_, free_);

  return *this;
}
```

hsaturn, 2023-02-25:
Ouch..... Thanks. Tiny string is not used yet because of this kind of bug. Thanks

closed 2023-02-27 by hsaturn