hsaturn/TinyMqtt #68: Version 0.9.12 was working but when recompiled throws errors in TinyString.cpp

Hwerow, 2023-03-12:
Error messages for str, size and free
exchange is not a member of std
namespace std has no member exchange

TinyString& TinyString::operator=(TinyString && other)
{
  if (this == &other) return *this;
  clear();
  str = std::exchange(other.str, str);
  size_ = std::exchange(other.size_, size_);
  free_ = std::exchange(other.free_, free_);
  return *this;
}

hsaturn, 2023-03-13:
Hello

Strange, I'm using TinyConsole and all is ok under platformio.

I know that TinyString has problems (and thus is not used).
Try to remove the file TinyString.cpp until I found a real solution.

If this does not work, you can safely remove the offending lines (unless you are using TinyString that is not at all recommended : unit tests: 
[image]

Thanks for the report

Hwerow, 2023-03-13:
Thanks Deleting TinyString.cpp now allows compilation with 0.9.19.

Thankyou for your program as I have found it very useful, and have just implemented a broker to broker arrangement

Cheers

closed 2023-03-13 by Hwerow