hsaturn/TinyMqtt #71: `mosquitto_pub` fails on 1.0.0 but works on <= 0.9.18

richievos, 2023-04-04:
I just updated to the 1.0.0 tag in an attempt to resolve some memory leaks I believe are occurring from TinyMqtt. After updating I can no longer connect to TinyMqtt.

Output under 0.9.18:

```
$ mosquitto_pub -d -h 192.168.0.70 -p 1883 -t foobar -m '{}'
Client null sending CONNECT
Client null received CONNACK (0)
Client null sending PUBLISH (d0, q0, r0, m1, 'foobar', ... (2 bytes))
Client null sending DISCONNECT
```

Output under 1.0.0:
```
$ mosquitto_pub -d -h 192.168.0.70 -p 1883 -t execute/measure_alk/manual/next_stepasdf -m '{}'
Client null sending CONNECT
Error: A network protocol error occurred when communicating with the broker.
```

I'll debug more, but posting in case others see connectivity issues in the meantime.

richievos, 2023-04-04:
Works in bf84e29. First stops working in e41452e.

Issue appears to be this [this line](https://github.com/hsaturn/TinyMqtt/commit/ead3ae90e9cee0369f854188e3e042625d7c49e0~2#diff-f6fc13eb02b890250382fe7305b6db7d46023337fb947569129455054b3fcef0R310):

```
// old
if (keep_alive && (millis() >= alive))
// new
if (keep_alive && (millis() >= alive - 5000))
```

what's that - 5000 about? Is that somethign accidentally left in after testing?

richievos, 2023-04-04:
I sent a PR removing that `-5000`: https://github.com/hsaturn/TinyMqtt/pull/72

I'm not sure what that is representing, so I'm not sure how to test ramifications. This version works for me though.

richievos, 2023-04-04:
Also unrelated, in the latest the apparent memory leak I was observing does appear fixed. That's promising.

I've been testing by sending an infinite loop of publishes with new clients to my esp:

```
$ while true
do
mosquitto_pub -h 192.168.0.70 -p 1883 -t foobar -m '{}'

done
```

Previously it'd eventually hang and stop responding. With 1.0.0 + my fix here it is working well.

dg6nee, 2023-04-04:
I vote a "+1", similar here, I created a fork quite a while ago and addressed the problem of memory leaks due to bad MQTT client handling (AFAIR pr23). This old version works on two of my devices, stable for many weeks.
I tried the latest official versions, but these still fail after less than one day. Same symptoms, device is still active (web server still works), but MQTT is dead, does not respond...

hsaturn, 2023-04-16:
PR about 5000ms merged