hsaturn/TinyMqtt #76: direct use of WiFiClient ?

real-bombinho, 2023-04-15:
Hello, I just forked your nice bit of work (thank you) and modified it to accommodate for W5500 LAN adaptors. Pretty much anything supported by the Ethernet library should work too. But I have not yet included other controllers than the ESP8266.
Thanks to you the adaption was fairly straight forward, mostly just needing a "using TcpClient = EthernetClient;" + server and some initialisation.

I noticed 2 instances of direct use of WiFiClient instead of TcpClient in TinyMqtt.cpp (lines 66 & 214 in the current main). Is there a particular reason?

Another problem I had, I could neither get the master or the later releases to work. They compiled fine but none of my clients connected to the broker. So I went back to the working release from the Arduino library manager, which I know to be working fine. I must admit, I did not spend a great deal of time yet on fault finding.

My main intention was to get it working on the W5500 first as I intended to port it to the RPi W5500-EVB-Pico at a later point.

hsaturn, 2023-04-16:
Hello, there is absolutely no reason.
Thanks for reporting this.

Best regards.

hsaturn, 2023-04-16:
Closed by https://github.com/hsaturn/TinyMqtt/commit/25602ec2f53d84e677b2f36a464849979ae69286

hsaturn, 2023-04-16:
Reopened because of this.

(Another problem I had, I could neither get the master or the later releases to work. They compiled fine but none of my clients connected to the broker. So I went back to the working release from the Arduino library manager, which I know to be working fine. I must admit, I did not spend a great deal of time yet on fault finding.) <= side note only

Could you provide more information please ?

real-bombinho, 2023-04-17:
I had used the Library manager for my first attempt and compilation worked fine and Clients connected straight away.
Then I replaced the libraries with the ones from the current main. Everything compiled fine but Clients did not connect to the broker. As I had seen main branches fail before I picked the latest release, 1.0.0. Same problem. Going back to the libraries from the Arduino library manager brought the clients straight back on.

In the mean time I noticed #71 , seemingly forcing [richievos](https://github.com/richievos) onto an older version too. I have not retried yet, but it would be worth trying to see if removing the -5000 would bring it live again.

hsaturn, 2023-04-17:
The -5000ms has been removed with https://github.com/hsaturn/TinyMqtt/pull/72.
The main branch should now be ok.
I don't know why arduino ide sticks to 0.9.6, but as this once was ok, this is better like that.

Thanks

closed 2023-04-17 by hsaturn