hsaturn/TinyMqtt #77: Message ID

real-bombinho, 2023-04-17:
Do you capture the message ID of transmitted messages? I could not find anything hinting on on a message ID, so I did a quick and dirty function to return the ID of a message.

real-bombinho, 2023-04-17:
I do have a problem with a client flooding with duplicates on QOS1, responding with PubAck solves the problem and keeps the network less choked.

hsaturn, 2023-04-17:
Qos is not and won't be handled. This is too memory consuming for a TinyLibrary.

real-bombinho, 2023-04-17:
From what I see QOS 1 is simple, you respond with SubAck and that is it.

hsaturn, 2023-04-17:
> 

Could you write a scenario with mosquitto for example.
I'm not sure, but I may reject Qos=1. I have to read the spec before around Qos=1.
Maybe aso I could patch as you did. But I cannot spend time on this so if you provide me enough information, this may be usefull (maybe mandatory) to ack.

real-bombinho, 2023-04-17:
The only other thing I am looking at right now is to resend to subscribers on QOS 0 rather passing 0 or 1 through yet still ignoring SubAck. This should keep the aim of supporting only QOS 0 on the right track and as far as I am aware it is absolutely in the right of the broker to resubmit at a different level than the subscriber subscribes at.

hsaturn, 2023-04-17:
Hmm. As I said, I have to read the spec, I guess there is something to do to increase TinyMqtt scope to what you are pointing.
Esecpially if modifications are light.

Yet, I have to fix more important issues.
I'm leaving this issue open. Feel free to add links to doc or every information that can help.

real-bombinho, 2023-04-17:
"In the publisher-to-subscriber delivery process, the publisher specifies the QoS level of a message in the PUBLISH packet. The broker typically forwards the message to the subscriber with the same QoS level. However, in some cases, the subscriber's requirements may necessitate a reduction in the QoS level of the forwarded message." 
[[source](https://www.emqx.com/en/blog/introduction-to-mqtt-qos)]

At the moment you pass right through, ignoring the QOS and thus someone sending at QOS 1 results in the client receiving a QOS 1 message too. This is ignoring the wishes of the client as meant to be.

real-bombinho, 2023-04-18:
I would not expect to increase the scope. Your aim is to not offer QOS. Hence making all outgoing publishes QOS 0 should be the better way rather passing through QOS 1. This sorts most issues with QOS without getting involved. The effort is to delete one bit on incoming. 

But since you pass through QOS 1 this generates a horrendous amount of additional traffic by continual resubmitting from clients using QOS 1, which can be eliminated by just responding a SubAck. No memory use, the ID is contained in buffer and just needs extraction. 

The submitting client does not know the QOS subscription level of the subscribing client. And therefore QOS 1 would (and should in accordance to your specification of ignoring QOS) only affect submissions to the broker.

real-bombinho, 2023-04-18:
I have submitted the SubAck.

I also have the code ready for dropping the QOS to 0, if you want to have a look at it.

hsaturn, 2023-04-20:
Hi, thanks for this report.

I agree with these modifications.
Is it possible to write unit tests before adding these (unit test can be ran under Linux by cloning AUnit EpoxyDuino and EspMock).
Without the unit test, further modifications have a lot of chance to remove this enhancement (as I do not check for Qos != 0 in my projects).

Best regards

real-bombinho, 2023-05-14:
I will close this issue in the next days as the issue itself is resolved.

I will open the dropQOS in a separate issue, once I have enough time to look at it. My DropQOS works fine with most clients but fails to disconnection on fragmented messages.

real-bombinho, 2023-05-15:
Closed as issue is resolved.

closed 2023-05-15 by real-bombinho