hsaturn/TinyMqtt #86: multiple retained messages for the same topic

real-bombinho, 2023-05-09:
TinyMqtt.cpp, line 965:

auto old = retained.find(topic);

does not work. Consequentially I get multiple retained messages for the same topic.

hsaturn, 2023-05-10:
Hello

I've modified the unit test on retained messages... Unable to reproduce this.

The test added is there https://github.com/hsaturn/TinyMqtt/commit/30e75b82e0461317339a4a7a94fbcd3268c0b136

Either you do not use the main branch, or .... the map is buggy on ESP implementation.

hsaturn, 2023-05-10:
Also this gave me the opportunity to add a test for the payload truncated message !!!

real-bombinho, 2023-05-10:
The std::map seems not to recognise any values unless I use hardcoded quoted strings.

hsaturn, 2023-05-10:
The commit for the test added is there (https://github.com/hsaturn/TinyMqtt/commit/30e75b82e0461317339a4a7a94fbcd3268c0b136)

hsaturn, 2023-05-10:
> The std::map seems not to recognise any values unless I use hardcoded quoted strings.

Yes, this sounds very strange.

real-bombinho, 2023-05-10:
Are you using the new 2.0.x environment?

hsaturn, 2023-05-10:
No. Unit test are not running on the ESP, but on Linux. This is why I suspect the map to be responsible of this :-(

cdconner16, 2023-11-26:
I can confirm the multiple messages for one topic with my ESP32 on the latest version of this library.
For the broker, I'm using simple-broker.ino with a couple modifications to the console and RETAIN = 10. Note that I do not fully understand MQTT yet so please correct me if I'm wrong about how it should work.

Last 16 publishes from my client with retain always set to true:
topic: thermostat/set_temp, payload: 71.00
topic: thermostat/set_temp, payload: 72.00
topic: thermostat/mode, payload: heat
topic: thermostat/relay_state_comp, payload: On
topic: thermostat/relay_state_rv, payload: On
topic: thermostat/state, payload: Heat
topic: thermostat/mode, payload: cool
topic: thermostat/mode, payload: off
topic: thermostat/relay_state_comp, payload: Off
topic: thermostat/relay_state_rv, payload: Off
topic: thermostat/state, payload: Off
topic: thermostat/mode, payload: heat
topic: thermostat/state, payload: Waiting_Comp_Off_Time
topic: thermostat/mode, payload: cool
topic: thermostat/mode, payload: off
topic: thermostat/state, payload: Off

After resetting my client and resubscribing to the topics I get the following output:
topic: thermostat/mode, payload: heat
topic: thermostat/mode, payload: cool
topic: thermostat/mode, payload: off
topic: thermostat/state, payload: Off
topic: thermostat/state, payload: Heat
topic: thermostat/state, payload: Off
topic: thermostat/relay_state_comp, payload: On
topic: thermostat/relay_state_comp, payload: Off
topic: thermostat/relay_state_rv, payload: On
topic: thermostat/relay_state_rv, payload: Off

However, I've lost the last value for the thermostat/set_temp topic because of the behavior. I would expect only the last value for each topic to be retained. I would also expect a topic to be completely removed from the buffer only if there are more than 10 topics that are trying to retain values.

cdconner16, 2023-11-26:
Trying to debug this because it's the last piece of code I need to make my system work.
The first problem is happening at line 977 in TinyMqtt.cpp, where retained.find(topic) isn't working for topics that already exist so we always take the if case and go into retainDrop().