hsaturn/TinyMqtt #94: Can not build simple-broker

hsaturn, 2023-09-10:
### Discussed in https://github.com/hsaturn/TinyMqtt/discussions/93

<div type='discussions-op-text'>

<sup>Originally posted by **DatutaX** September  9, 2023</sup>
i try to compile the simple-broker with Arduino ide 2.2.1 but get error.
 #include <vector>
          ^~~~~~~~
compilation terminated.
exit status 1
Compilation error: exit status 1

how to compile this ?
Regards</div>

hsaturn, 2023-09-10:
Hello

I've tried to build the broker with IDE 2.2.1, TinyMqtt 0.9.6 (the one reachable from IDE yet) with no problem.

Could you provide more information.
Also I've relase the 1.1.0 which now may be visible from the Arduino IDE 2 (maybe in 24h, I've just released).

Regards.

real-bombinho, 2023-10-29:
I currently have a number of issues with [ESP8266 Community version 3.1.2] rather than with the IDE. Using older versions might be worth a try.

hsaturn, 2023-11-04:
Hello, you have to mention what version of TinyMqtt you are using. I cannot help without this.

ty8447, 2023-12-04:
I am also having this issue. I have TinyConsole 0.4.7 and TinyMQTT 1.1.0 (which appears as 1.0.1 in the IDE) and Arduino IDE 2.2.1. When compiling the code I get this error

C:\Users\XXX\AppData\Local\Temp\.arduinoIDE-unsaved2023113-16176-quug4r.rc83h\simple-broker\simple-broker.ino: In function 'void setup()':
C:\Users\XXX\AppData\Local\Temp\.arduinoIDE-unsaved2023113-16176-quug4r.rc83h\simple-broker\simple-broker.ino:33:27: error: 'red' is not a member of 'TinyConsole'
   Console << TinyConsole::red << "****** PLEASE MODIFY ssid/password *************" << endl;
                           ^~~
C:\Users\XXX\AppData\Local\Temp\.arduinoIDE-unsaved2023113-16176-quug4r.rc83h\simple-broker\simple-broker.ino:42:27: error: 'green' is not a member of 'TinyConsole'
   Console << TinyConsole::green << "Connected to " << ssid << "IP address: " << WiFi.localIP() << endl;
                           ^~~~~

exit status 1

Compilation error: 'red' is not a member of 'TinyConsole'

Where the example code I am using is the following

`#include "TinyMqtt.h"   // https://github.com/hsaturn/TinyMqtt

const uint16_t PORT = 1883;
const uint8_t  RETAIN = 10;  // Max retained messages

MqttBroker broker(PORT, RETAIN);

/** Basic Mqtt Broker
  *
  *  +-----------------------------+
  *  | ESP                         |
  *  |       +--------+            |
  *  |       | broker |            | 1883 <--- External client/s
  *  |       +--------+            |
  *  |                             |
  *  +-----------------------------+
  *
  *  Your ESP will become a MqttBroker.
	*  You can test it with any client such as mqtt-spy for example
	*
	* Messages are retained *only* if retain > 0
	*
  */

const char* ssid = "Danny";
const char* password = "Dev1to!!";

void setup()
{
  Serial.begin(115200);

	if (strlen(ssid)==0)
		Console << TinyConsole::red << "****** PLEASE MODIFY ssid/password *************" << endl;

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    Serial << '.';
    delay(500);
  }
  Console << TinyConsole::green << "Connected to " << ssid << "IP address: " << WiFi.localIP() << endl;

  broker.begin();
  Console << "Broker ready : " << WiFi.localIP() << " on port " << PORT << endl;
}

void loop()
{
  broker.loop();
}
`