hsaturn/TinyMqtt #100: Disconnections Management

Leonidas00000, 2024-02-01:
I'm trying to implement a broker with this library and i wanted to know if there is a easy way to detect disconnected clients, i didn't found "onDisconnect()" event or anything like that, i want to know how can i detect a disconnected client and get its "id".

hsaturn, 2024-03-06:
Hello

You're right, this is on my todo list. I've never been disconnected so I did not implement this. Not planned but I keep this issue for when I have time.

Aurentiaco35, 2024-05-30:
Hello,

I also want to detect disconnected clients, do you have any process in that area?

Leonidas00000, 2024-05-30:
Hello, 
even though there is no a disconnection management implemented, i managed to do it, i don´t know if this helps you but, i detected a client disconnection using "Clients_connected = broker.clientsCount();"

Aurentiaco35, 2024-06-03:
That would help thank you so much.