hsaturn/TinyMqtt #103: simple-broker not compiling

olleyOop, 2024-08-22:
Hello,

after I updated the library in Arduino IDE 2.3.2 I can't compile my project anymore.
Even if I try to compile the example "simple-broker" I get the following error:

> In file included from c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyConsole.h:5,
>                  from c:\Users\oreic\Documents\Arduino\libraries\TinyMqtt\src/StringIndexer.h:6,
>                  from c:\Users\oreic\Documents\Arduino\libraries\TinyMqtt\src/TinyMqtt.h:39,
>                  from C:\Users\oreic\AppData\Local\Temp\.arduinoIDE-unsaved2024722-3888-hvfh85.6dmcg\simple-broker\simple-broker.ino:1:
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h: In member function 'TinyTerm::CallBackKey TinyTerm::onKey(CallBackKey)':
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h:95:56: error: call of overloaded 'exchange(TinyTerm::CallBackKey&, TinyTerm::CallBackKey&)' is ambiguous
>    95 |     CallBackKey onKey(CallBackKey cb) { return exchange(callback_key, cb); }
>       |                                                ~~~~~~~~^~~~~~~~~~~~~~~~~~
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h:12:3: note: candidate: 'T exchange(T&, U&&) [with T = std::function<void(TinyTerm::KeyCode)>; U = std::function<void(TinyTerm::KeyCode)>&]'
>    12 | T exchange(T& obj, U&& new_value)
>       |   ^~~~~~~~
> In file included from c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h:6:
> c:\users\oreic\appdata\local\arduino15\packages\esp32\tools\esp-x32\2302\xtensa-esp32-elf\include\c++\12.2.0\utility:93:5: note: candidate: 'constexpr _Tp std::exchange(_Tp&, _Up&&) [with _Tp = function<void(TinyTerm::KeyCode)>; _Up = function<void(TinyTerm::KeyCode)>&]'
>    93 |     exchange(_Tp& __obj, _Up&& __new_val)
>       |     ^~~~~~~~
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h: In member function 'TinyTerm::CallBackMouse TinyTerm::onMouse(CallBackMouse)':
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h:96:80: error: call of overloaded 'exchange(TinyTerm::CallBackMouse&, TinyTerm::CallBackMouse&)' is ambiguous
>    96 |     CallBackMouse onMouse(CallBackMouse cb) { mouseTrack(true); return exchange(callback_mouse, cb); }
>       |                                                                        ~~~~~~~~^~~~~~~~~~~~~~~~~~~~
> c:\Users\oreic\Documents\Arduino\libraries\TinyConsole\src/TinyTerm.h:12:3: note: candidate: 'T exchange(T&, U&&) [with T = std::function<void(const TinyTerm::MouseEvent&)>; U = std::function<void(const TinyTerm::MouseEvent&)>&]'
>    12 | T exchange(T& obj, U&& new_value)
>       |   ^~~~~~~~
> c:\users\oreic\appdata\local\arduino15\packages\esp32\tools\esp-x32\2302\xtensa-esp32-elf\include\c++\12.2.0\utility:93:5: note: candidate: 'constexpr _Tp std::exchange(_Tp&, _Up&&) [with _Tp = function<void(const TinyTerm::MouseEvent&)>; _Up = function<void(const TinyTerm::MouseEvent&)>&]'
>    93 |     exchange(_Tp& __obj, _Up&& __new_val)
>       |     ^~~~~~~~
> C:\Users\oreic\AppData\Local\Temp\.arduinoIDE-unsaved2024722-3888-hvfh85.6dmcg\simple-broker\simple-broker.ino: In function 'void setup()':
> C:\Users\oreic\AppData\Local\Temp\.arduinoIDE-unsaved2024722-3888-hvfh85.6dmcg\simple-broker\simple-broker.ino:29:41: error: 'red' is not a member of 'TinyConsole'
>    29 |                 Console << TinyConsole::red << "****** PLEASE MODIFY ssid/password *************" << endl;
>       |                                         ^~~
> C:\Users\oreic\AppData\Local\Temp\.arduinoIDE-unsaved2024722-3888-hvfh85.6dmcg\simple-broker\simple-broker.ino:38:27: error: 'green' is not a member of 'TinyConsole'
>    38 |   Console << TinyConsole::green << "Connected to " << ssid << "IP address: " << WiFi.localIP() << endl;
>       |                           ^~~~~
> 
> exit status 1
> 
> Compilation error: 'red' is not a member of 'TinyConsole'

Any help would be appreciated.

papercodeIN, 2024-09-02:
Same error for me as well any solution for this problem ?

Tschmidt9490, 2024-09-07:
go in to the tinyconsole library and switch this 

TinyString& TinyString::operator=(TinyString && other)
{
  if (this == &other) return *this;
  clear();
  str = std::exchange(other.str, str);
  size_ = std::exchange(other.size_, size_);
  free_ = std::exchange(other.free_, free_);
  return *this;
}

With this 

TinyString& TinyString::operator=(TinyString && other)
{
  if (this == &other) return *this;
  clear();
  std::swap(other.str, str);
  std::swap(other.size_, size_);
  std::swap(other.free_, free_);
  return *this;
}

papercodeIN, 2024-09-07:
> go in to the tinyconsole library and switch this
>
> TinyString& TinyString::operator=(TinyString && other) { if (this == &other) return *this; clear(); str = std::exchange(other.str, str); size_ = [...]

Thank You friend, also can you help one more thing ? Could you please tell the file name ? Really sorry for silly question.

Tschmidt9490, 2024-09-07:
Tinystring.cpp

lutorm, 2024-10-30:
I'm getting the same error, but my Tinystring.cpp already uses std::swap. Besides, the error above says that the call in TinyTerm::onKey is ambiguous. If I change "exchange" to "::exchange" to force the use of the template at the top of the file, the ambiguity is broken. But then I'm getting another error:

```cpp
/tmp/.arduinoIDE-unsaved2024930-2070929-yr320l.vzzcp/simple-broker/simple-broker.ino: In function 'void setup()':
/tmp/.arduinoIDE-unsaved2024930-2070929-yr320l.vzzcp/simple-broker/simple-broker.ino:29:41: error: 'red' is not a member of 'TinyConsole'
   29 |                 Console << TinyConsole::red << "****** PLEASE MODIFY ssid/password *************" << endl;
      |                                         ^~~
/tmp/.arduinoIDE-unsaved2024930-2070929-yr320l.vzzcp/simple-broker/simple-broker.ino:38:27: error: 'green' is not a member of 'TinyConsole'
   38 |   Console << TinyConsole::green << "Connected to " << ssid << "IP address: " << WiFi.localIP() << endl;
```
and indeed TinyConsole does not have any such members...

dkrape, 2024-12-31:
I am getting the same errors. Using Arduino IDE version 2.3.4 and 0.9.16 of the TinyMqtt library.